stickbreaker/OneWire #1: OneWire library and Arduino

aqucomp, 2018-07-10:
Hi,
after I made update on Arduino to ESP32 through BoardManager https://desire.giesecke.tk/index.php/2018/07/01/add-esp32-to-boards-manager-workaround/
and manually deleted git from Arduino environment
the DallasTemperature library dont work. https://github.com/milesburton/Arduino-Temperature-Control-Library
I had installed your OneWire library before and i think it can be sth with that.
I didn't change the hardware. Is now the OneWire library part of Arduino-ESP32 ?

```
#include <DallasTemperature.h>

//DS18B20 Temperature Sensor
#define ONE_WIRE_BUS 25 

/********************************************************************/
// Setup a oneWire instance to communicate with any OneWire devices  
// (not just Maxim/Dallas temperature ICs) 
OneWire oneWire(ONE_WIRE_BUS); 
/********************************************************************/
// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; 
  }

  Serial.println("DS 18B20 Begin.");
  sensors.begin(); 
  Serial.print("Device Number: ");
  Serial.println(sensors.getDeviceCount());
  Serial.print("Dallas Number: ");
  Serial.println(sensors.getDS18Count());
}
```
The answear is:
```
DS 18B20 Begin.
Device Number: 0
Dallas Number: 0

```
Thx.

stickbreaker, 2018-07-10:
@aqucomp my modified OneWire works with ESP32, the original author and maintainer is Paul Stoffregen @PaulStoffregen.  [oneWire](http://www.pjrc.com/teensy/td_libs_OneWire.html).  I propose these changes but they were not accepted, he chose a different coding style, I was not able to use his new fix under Arduino-esp32.  My modifications work for me.

I don't know the 'Official' status of `OneWire()` on ESP32 from Paul.  When I spoke with him he was not interested in supporting the ESP32. His interest trajectory is with the Teensy path on Arduino.

Chuck.

aqucomp, 2018-07-10:
I had probably problem with my environment.
I deleted my old onewire lib and install it again and it work :)
Thank you.
But why you don't integrate your library into "standard" arduino-esp32? (If Paul don't want it.)
Andrew.

stickbreaker, 2018-07-10:
@aqucomp I don't control Arduino-ESP32.  

@me-no-dev is the controlling influence for Arduino-ESP32.  I don't know what ramifications of adding this library are.  I do know that BLE and AzureIot are submodules. 

Possibly if there is enough interest in my OneWire branch, it might be included as a submodule. I have no control of this 

Chuck.

aqucomp, 2018-07-10:
ok, thank you :)
Andrew.

closed 2018-07-10 by aqucomp

me-no-dev, 2018-07-10:
@stickbreaker since it's your library, it's up to you to propose/pr it 😛

me-no-dev, 2018-07-10:
BTW TBH OneWire should be implemented in RMT on ESP32 (there is a PR for RMT).

PaulStoffregen, 2018-07-10:
> I don't know the 'Official' status of OneWire() on ESP32 from Paul.

It works.  I personally tested a DS18B20 sensor on an ESP32 board.

stickbreaker, 2018-07-10:
@me-no-dev Why in RMT, isn't RMT the infra red communication with a 30kHz to 60kHz carrier?

@paulstoffregen The only answer I can give you, is when people have issues using your code with the ESP32, they come here. 
In April, 2018 @everslick found and fixed a protocol timing violation cause by the ESP32 ICACHE. `read_bit()` and `write_bit()`  not being resident in IRAM (always present in physical RAM).  An ICACHE miss delayed the operation enough to violate the protocol timing. This would only show up in a test case that caused ICACHE flushing.  
Here is the commit [fix unstable timing](https://github.com/stickbreaker/OneWire/commit/79a534e80371313e3db287f725e3bf3db10f3ead)

Chuck.

me-no-dev, 2018-07-11:
RMT is a very versatile peripheral that can be used for all kinds of weird signaling. You tell for each puls e how long it should be low and how long high. Same way it returns that data when reading. Point is you do not need to delay or use pin interrupts to send/receive data from OneWire.

stickbreaker, 2018-07-11:
@me-no-dev when I get a chance, I'll look at RMT.  I don't expect that to happen for a few months though.

Chuck.

hattesen, 2019-09-22:
Shouldn't a separate issue be created for the "RMT support on ESP32" investigation/implementation? Currently, the "task" is only referenced in an unrelated, and closed, issue (this)

stickbreaker, 2019-09-22:
@hattesen There is a RMT library that works with IR transmit/receive. I looked at the RMT hardware but did not see a way to implement the OneWire protocol with it.  OneWire is an "and" protocol with bit stretching.  The slave drives the shared line AFTER the Master sends a start bit.  The RMT hardware is designed to send/receive bit streams. 

A Pull from my fork of Paul Stoffregen's OneWire was not accepted.  Paul did not like my modification style.  It works for me, It is not worth my effort to conform to Paul.

Chuck.