stickbreaker/OneWire #2: compiling for both ESP8266 and ESP32 with OneWire Libraries

bill-orange, 2019-05-05:
First off, this OneWire library works very well.  The CRC errors occurring every minute or so disappeared.  Nice.   I am however having a problem integrating this library into my Arduino IDE cleanly.  This may be a bit off topic since the library works well, but I suspect that I am not alone in encountering this problem.

Installing this library as a Zip using the board manager creates a folder OneWire-master.  The existing library is in a folder named OneWire .  In order to make the compiler pick your library the OneWire folder has to be deleted.  

Unfortunately the OneWire library in OneWire-master will not compile with an ESP8266.  I have both ESP8266 and ESP32 boards with code needing OneWire.  Restoring the OneWire Folder every time  want to compile for an ESP8266 is awkward at best.  

I tried placing the .ccp file and .h file in the folder with the sketch and using #include "OneWire.h" rather than #include <OneWire.h>.  That generated compiler errors when OneWire could no longer find external components.

I tried renaming OneWire and that also generated compiler errors even with the reference to the .h file within the .cpp revised.

There must be some sort of slick way to handle this.  

Advice?

stickbreaker, 2019-05-05:
Bill, the default Zip generation creates that "OneWire-Master" directory name..

If you wanted to create a merge with the ESP8266 code with #ifdef's I'm not adverse.  

I don't Use ESP8266's so you would have to do all the work and testing.  I just use it with ESP32.  I tried to get Paul to accept a merge, but, he wanted the coding style to be something I couldn't support. I did not believe that his design would be beneficial.  This repo is optimized for ESP32, It serves my purposes and others have found it useful.  

Bill, I just generated a "pre-Release" try grabbing that release zip and see if it works for you.

Chuck.

bill-orange, 2019-05-06:
Yes! In my exhaustive 15 minutes of testing, it works!  

I tried to install the zip off of github and got the message 'not a valid zip library'.  Undeterred, I unzipped the OneWire-master folder and substituted it in my library folder.  

Compile generated a warning in both the ESP8266 compilation and the ESP32 compilation.  Odd that it would do it on both.

WARNING: library OneWire-master claims to run on (ESP32) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s).

                                                             or

WARNING: library OneWire-master claims to run on (ESP32) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).

It was only a warning and compilation was successful.  In my brief testing, I am getting good data on both boards.  On the ESP32 board I have found it necessary to keep the Dallas task pinned to one core even with your code to avoid the occasional -195 reading.  That's not a big deal.

I have a sketch that generates a short term weather forecast by using OpenWeatherMap forecasts combined with the change in measured temperature over the previous hour to create a graph of the likely upcoming temperature (ie. should you put your jacket in the car).  This was a request of my friend Walt, thus the title on the graph.

[](http://orangecaweather.duckdns.org:8021/)

Thanks.  Your work could help a lot of people.  I cannot be the only one running both boards.

Bill

stickbreaker, 2019-05-06:
looks like a Capitalization error  "ESP32" instead of "esp32"

You grabbed the zip from the Release section right?

bill-orange, 2019-05-06:
Not sure. I only saw master. I did not see release. It had your changes from yesterday. 

Code on both the ESP8266 and ESP32 fine overnight.