stickbreaker/OneWire #3: Debugging failing transactions

stickbreaker, 2019-11-17:
Continuation of issue with [Arduino-Esp32#1335](https://github.com/espressif/arduino-esp32/issues/1335) with @OmarAlkassab .

OmarAlkassab, 2019-11-17:
Dear @stickbreaker, actually I can’t go to the laboratory to make the test with the logic analyzer, I am actually in home due to some problems in my country. I noticed now that the task hangs after some hours! The task hangs when the task is moved to the core 1.

stickbreaker, 2019-11-18:
@OmarAlkassab Well, when you have some time to build an example that replicates the problem I'll take a look. Until I have something that duplicates the problem I can't try fixing it.

Chuck.

stickbreaker, 2020-01-19:
Closed due to inactivity.

closed 2020-01-19 by stickbreaker