stickbreaker/OneWire #4: Warning messages during compile

vbandke, 2020-01-16:
I am using this library in PlatformIo, and compile a program with DS18B20 sensor support.  The program works nicely, but during compile I get a few messages referring to redefines.  These are warnings, but I would like those to be gone.  What needs to  be done ?

Here are the messages:
```

.pio\libdeps\release\OneWire\OneWire.cpp:156:0: warning: "noInterrupts" redefined
 #define noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
 ^
In file included from .pio\libdeps\release\OneWire\OneWire.h:11:0,
                 from .pio\libdeps\release\OneWire\OneWire.cpp:153:
C:\users\volker\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:79:0: note: this is the location of the previous definition
 #define noInterrupts() cli()
 ^
.pio\libdeps\release\OneWire\OneWire.cpp:157:0: warning: "interrupts" redefined
 #define interrupts() portEXIT_CRITICAL(&mux);}
 ^
In file included from .pio\libdeps\release\OneWire\OneWire.h:11:0,
                 from .pio\libdeps\release\OneWire\OneWire.cpp:153:
C:\users\volker\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:78:0: note: this is the location of the previous definition
 #define interrupts() sei()
 ^
.pio\libdeps\release\OneWire\OneWire.cpp:621:20: warning: extra tokens at end of #undef directive
 #undef noInterrupts()
                    ^
.pio\libdeps\release\OneWire\OneWire.cpp:622:18: warning: extra tokens at end of #undef directive
 #undef interrupts()
                  ^
In file included from src\main.cpp:12:0:
.pio/libdeps/release/OneWire/OneWire.h:246:2: warning: #warning "ESP32 OneWire testing" [-Wcpp]
 #warning "ESP32 OneWire testing"
  ^
In file included from .pio\libdeps\release\DallasTemperature_ID54\DallasTemperature.h:22:0,
                 from .pio\libdeps\release\DallasTemperature_ID54\DallasTemperature.cpp:6:
.pio\libdeps\release\OneWire/OneWire.h:246:2: warning: #warning "ESP32 OneWire testing" [-Wcpp]
 #warning "ESP32 OneWire testing"
  ^
.pio\libdeps\release\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::reset()':
.pio\libdeps\release\OneWire\OneWire.cpp:184:27: warning: unused variable 'reg' [-Wunused-variable]
     volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                           ^
.pio\libdeps\release\OneWire\OneWire.cpp: In member function 'void OneWire::write_bit(uint8_t)':
.pio\libdeps\release\OneWire\OneWire.cpp:219:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^
.pio\libdeps\release\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::read_bit()':
.pio\libdeps\release\OneWire\OneWire.cpp:251:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
```

with best regards

Volker Bandke

stickbreaker, 2020-01-19:
Not all of them can be addressed. These are telling you the truth.  
In `Arduino.h` `#noInterrupts()` is defined as `cli()`. But, that does not work for OneWire.  it is changed to `#define noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)` 
```
.pio\libdeps\release\OneWire\OneWire.cpp:156:0: warning: "noInterrupts" redefined
 #define noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
 ^
In file included from .pio\libdeps\release\OneWire\OneWire.h:11:0,
                 from .pio\libdeps\release\OneWire\OneWire.cpp:153:
C:\users\volker\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:79:0: note: this is the location of the previous definition
 #define noInterrupts() cli()
 ^
.pio\libdeps\release\OneWire\OneWire.cpp:157:0: warning: "interrupts" redefined
 #define interrupts() portEXIT_CRITICAL(&mux);}
 ^
In file included from .pio\libdeps\release\OneWire\OneWire.h:11:0,
                 from .pio\libdeps\release\OneWire\OneWire.cpp:153:
C:\users\volker\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:78:0: note: this is the location of the previous definition
 #define interrupts() sei()
 ^

.pio\libdeps\release\OneWire\OneWire.cpp:621:20: warning: extra tokens at end of #undef directive
 #undef noInterrupts()
                    ^
.pio\libdeps\release\OneWire\OneWire.cpp:622:18: warning: extra tokens at end of #undef directive
 #undef interrupts()
                  ^
```
These two can be fixed by commenting them out or deleting them entirely.
```
In file included from src\main.cpp:12:0:
.pio/libdeps/release/OneWire/OneWire.h:246:2: warning: #warning "ESP32 OneWire testing" [-Wcpp]
 #warning "ESP32 OneWire testing"
  ^
In file included from .pio\libdeps\release\DallasTemperature_ID54\DallasTemperature.h:22:0,
                 from .pio\libdeps\release\DallasTemperature_ID54\DallasTemperature.cpp:6:
.pio\libdeps\release\OneWire/OneWire.h:246:2: warning: #warning "ESP32 OneWire testing" [-Wcpp]
 #warning "ESP32 OneWire testing"
  ^
```
These I don't know, I'll have to try to understand why the compiler is generating the warning. This OneWire library was originally designed for AVR processors.  It used direct hardware access, I just modified it enough to work on the ESP32:
```
.pio\libdeps\release\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::reset()':
.pio\libdeps\release\OneWire\OneWire.cpp:184:27: warning: unused variable 'reg' [-Wunused-variable]
     volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                           ^
.pio\libdeps\release\OneWire\OneWire.cpp: In member function 'void OneWire::write_bit(uint8_t)':
.pio\libdeps\release\OneWire\OneWire.cpp:219:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^
.pio\libdeps\release\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::read_bit()':
.pio\libdeps\release\OneWire\OneWire.cpp:251:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
```

Chuck.

vbandke, 2020-01-19:
Thanks for taking the time responding so quickly.

As these warnings are not showstoppers in any way, don‘t spend too much time investigating.  As of the redefines - can‘t those msgs be obliterated by just putting an #undef in front of the (re-)#define?

Best regards from Germany

Volker

stickbreaker, 2020-01-19:
Try it. If it work you could upload a pull request, and fix it for everyone.

Chuck.