stickbreaker/OneWire #5: platformio & constructor

ramarro123, 2020-02-10:
hello,

do you plan to release it on platform.io library manager?

and, one more question, why did you change the constructor form original ds18b20?

i used to have an array of OneWire in a class, and construct the object when i got an mqtt message (after the class been loaded) but since there is no empty constructor i can't for now.

on original version i just used the empty constructor and then reinitialize it

stickbreaker, 2020-02-10:
>do you plan to release it on platform.io library manager?

I don't use PlatformIO so I have no interest in building a library manager.

>and, one more question, why did you change the constructor form original ds18b20?

I do not believe I changed the constructor. The only differences are how critical sections are coded.

>i used to have an array of OneWire in a class, and construct the object when i got an mqtt message (after the class been loaded) but since there is no empty constructor i can't for now.
on original version i just used the empty constructor and then reinitialize it

I don't know what you are trying to say here? You can do a compare with Paul Stoffregen see all the changes I made.  I don't see any changes to the Class Constructor?

Chuck.

ramarro123, 2020-02-10:
version of paul
```
  public:
    OneWire() { }
    OneWire(uint8_t pin) { begin(pin); }
    void begin(uint8_t pin);

```
your version

```
  public:
    OneWire( uint8_t pin);

```

probably you forked a previous version of paul codebase? you plan to merge back with their change?

stickbreaker, 2020-02-10:
I submitted these changes to Paul back in 2017, but he did not like the change style. 

So, I doubt a new merge would be accepted. 

I forked this Dec 2017, so those construct changes done March 2018 and June 2019 aren't included. 

>probably you forked a previous version of paul codebase? you plan to merge back with their change?

I'd have to test the changes, but, don't have time currently. so I don't know when, if ever I would update it.  It is working for me.

Chuck

pbolduc, 2020-09-22:
> do you plan to release it on platform.io library manager?

There should be no need to release it on the platform io library manager. Consumers can just use git reference, see [Library options](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps)

You can add a direct reference to the latest, specific version in your platformio.ini. Here I reference the specific tagged version

```
lib_deps = ${env.lib_deps}
    milesburton/DallasTemperature@^3.9.0
    https://github.com/stickbreaker/OneWire.git#v2.3.3-ESP32
```